yoursunny/WifiEspNow #1: error: 'ESP_NOW_ROLE_COMBO' not declared in this scope?

jbowser21, 2018-01-27:
I wanted to try the examples and for some reason they won't compile and give the following error:

error: 'ESP_NOW_ROLE_COMBO' was not declared in this scope

          esp_now_set_self_role(ESP_NOW_ROLE_COMBO) == 0 &&

I've tried a few different boards and all give the same error...

yoursunny, 2018-02-02:
What version (commit hash) of https://github.com/esp8266/Arduino are you running? This library is tested recently with git master branch. It may not be compatible with an older release version.

yoursunny, 2018-05-16:
Author did not post an update in past 90 days so I’m closing this issue. If issue persists, please post a comment and I’ll reopen.

closed 2018-05-16 by yoursunny