yoursunny/WifiEspNow #2: Weird behaviour with ESP-Now

bertrik, 2018-05-16:
Maybe an overly general question/remark, but how is ESP-Now working for you?
I do still see a bunch of weird behaviours.

Examples:
* I can not really get ESP-Now to work reliably on other channels than channel 1. Behaviour I'm seeing for example: message doesn't arrive at all, message arrives only sometimes, message arrives but sender does not receive an ACK.
* If I listen to one channel (set up an inactive AP + ESP-NOW on the receiver and check the frequency on my phone) and transmit on another (by adding the peer MAC + channel), my messages usually still arrive!
* even if I don't add a peer on the receiver side, the callback still gets called. I would have expected it only to work after adding a broadcast peer.

On channel 1 it works quite well for me, messages seem to have long range too,  I can send from any place in the house to the receiver node and still receive the ACK. It allows me to wake up from sleep (or just power up), send the message and be done in about 200 ms. The project I want to use this in, needs low-latency transmitting and low power consumption (battery powered), which ESP-Now seems very suitable for. My project page: https://revspace.nl/EspNowSkip

yoursunny, 2018-05-16:
> I can not really get ESP-Now to work reliably on other channels than channel 1. Behaviour I'm seeing for example: message doesn't arrive at all, message arrives only sometimes, message arrives but sender does not receive an ACK.

This is an issue of underlying firmware, and not caused by **WifiEspNow** library.

> If I listen to one channel (set up an inactive AP + ESP-NOW on the receiver and check the frequency on my phone) and transmit on another (by adding the peer MAC + channel), my messages usually still arrive!

This is a feature of ESP-NOW protocol. The transmitter will change its own frequency temporarily and transmit on the recipient’s channel.

> even if I don't add a peer on the receiver side, the callback still gets called. I would have expected it only to work after adding a broadcast peer.

I’ve never seen `RxCallback` invoked without anyone transmitting. Please provide exact steps to reproduce.

yoursunny, 2018-05-17:
> What I mean with the channel mismatch is that messages still arrive, even when they are sent to the wrong channel!

This is a behavior of underlying firmware. I guess `channel` is just a hint.

> I didn't mean that the RxCallback is invoked without transmission, but I mean it is invoked even when no peer is set up at all on the receiving side.

`RxCallback` does not require a registered peer. It is invoked whenever a message arrives at the current node.
https://github.com/yoursunny/WifiEspNow/blob/1e4122a4dff8be4b363e52772b95b99e5dc78e33/src/WifiEspNow.cpp#L126-L132

bertrik, 2018-06-21:
Thanks for the responses

closed 2018-06-21 by bertrik