yoursunny/WifiEspNow #5: Is it possible to enable encryption?

untitledlt, 2019-01-23:
Can't find anything in examples.
Thanks!

yoursunny, 2019-01-23:
`WifiEspNow.addPeer` has an `key` parameter that accepts an encryption key.
Fair warning: I’ve never used it and don’t know how strong the encryption scheme is.

coelner, 2021-01-16:
https://www.espressif.com/sites/default/files/documentation/esp-now_user_guide_en.pdf pg.4  (but maybe this is outdated)
> Limited encrypted peers. 10 encrypted peers at the most are supported in Station mode; 6 at the most in SoftAP or SoftAP + Station mode. Multiple unencrypted peers are supported, however, their total number should be less than 20, including encrypted peers.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_now.html#security

> ESP-NOW uses the CCMP method, which is described in IEEE Std. 802.11-2012, to protect the vendor-specific action frame. The Wi-Fi device maintains a Primary Master Key (PMK) and several Local Master Keys (LMK). The lengths of both PMK and LMk are 16 bytes. 
> PMK is used to encrypt LMK with the AES-128 algorithm. Call esp_now_set_pmk() to set PMK. If PMK is not set, a default PMK will be used.
> LMK of the paired device is used to encrypt the vendor-specific action frame with the CCMP method. The maximum number of different LMKs is six. If the LMK of the paired device is not set, the vendor-specific action frame will not be encrypted.
> Encrypting multicast vendor-specific action frame is not supported.

This library does not pass any key (this would be a LMK):
https://github.com/yoursunny/WifiEspNow/blob/b5d96e75a65504b1302a42c17634493f7e90fddb/src/WifiEspNowBroadcast.cpp#L173

I suggest a function to pass through a user defined PMK (aka KOK) to reach this call:
https://github.com/esp8266/Arduino/blob/74819a763bfb6e9890a57411dcea4aba221a778d/tools/sdk/include/espnow.h#L76
or/and this:
https://github.com/espressif/arduino-esp32/blob/8d0e68db4f73c6243be4e9c5955ef1eb842dd95b/tools/sdk/include/esp32/esp_now.h#L303
and add a `setLMK` function for the broadcast implementation.

I'm not sure if I sum it up right:
 - max 20 peers at all for ESPNOW
 - max 6 different LMK , which would allow 20 encrypted peers with a single LMK?
 - at least one LMK is needed

### related
https://github.com/espressif/ESP8266_NONOS_SDK/issues/114
https://github.com/espressif/ESP8266_NONOS_SDK/issues/8
https://github.com/esp8266/Arduino/issues/6174

closed 2021-08-21 by yoursunny