yoursunny/WifiEspNow #10: On ESP32 system crashes when esp_now_xxx functions called without successful begin()

TD-er, 2020-09-30:
Crashes like these occur if any WiFiEspNow function (other then `begin()` ) is called without a successful call to `begin()` (or not called at all)

```
Core 1 register dump:
PC      : 0x401bb499  PS      : 0x00060030  A0      : 0x800d7180  A1      : 0x3ffb1d90  
A2      : 0x00000000  A3      : 0x3ffbfb30  A4      : 0x3ffca3b4  A5      : 0x3ffb259c
A6      : 0x000000ff  A7      : 0x000000ff  A8      : 0x80083782  A9      : 0x3ffb1d70  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000002  A13     : 0x00000202  
A14     : 0x3ffb25c4  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000003c  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

Backtrace: 0x401bb499:0x3ffb1d90 0x400d717d:0x3ffb1db0 0x4012e305:0x3ffb1dd0 0x4012e415:0x3ffb1e00 0x40137af1:0x3ffb1e20 0x40137b35:0x3ffb1e60 0x400fed44:0x3ffb1e90 0x4013d48f:0x3ffb1ec0 0x4013ef7a:0x3ffb1f20 0x4013f905:0x3ffb1f40 0x4011cd7e:0x3ffb1f80 0x4014f631:0x3ffb1fb0 0x40089d59:0x3ffb1fd0
  #0  0x401bb499:0x3ffb1d90 in esp_now_is_peer_exist at ??:?
  #1  0x400d717d:0x3ffb1db0 in WifiEspNowClass::hasPeer(unsigned char const*) const at .pio\libdeps\custom_ESP32_4M316k_ETH\WifiEspNow\src/WifiEspNow.cpp:68
```

My fix for this is to simply add a bool member `bool m_begin = false;` and only set it to `true` on a successful `begin()`:

```c++
bool
WifiEspNowClass::begin()
{
  m_begin = esp_now_init() == 0 &&
#ifdef ESP8266
         esp_now_set_self_role(ESP_NOW_ROLE_COMBO) == 0 &&
#endif
         esp_now_register_recv_cb(reinterpret_cast<esp_now_recv_cb_t>(WifiEspNowClass::rx)) == 0 &&
         esp_now_register_send_cb(reinterpret_cast<esp_now_send_cb_t>(WifiEspNowClass::tx)) == 0;
  return m_begin;
}
```

Then just check this `m_begin` flag on any function that needs to call `esp_now_xxx` functions:
```c++

void
WifiEspNowClass::end()
{
  if (m_begin) {
    esp_now_deinit();
    m_begin = false;
  }
}

int
WifiEspNowClass::listPeers(WifiEspNowPeerInfo* peers, int maxPeers) const
{
  int n = 0;
  if (m_begin) {
#if defined(ESP8266)
    for (u8* mac = esp_now_fetch_peer(true);
        mac != nullptr;
        mac = esp_now_fetch_peer(false)) {
      uint8_t channel = static_cast<uint8_t>(esp_now_get_peer_channel(mac));
#elif defined(ESP32)
    esp_now_peer_info_t peer;
    for (esp_err_t e = esp_now_fetch_peer(true, &peer);
        e == ESP_OK;
        e = esp_now_fetch_peer(false, &peer)) {
      uint8_t* mac = peer.peer_addr;
      uint8_t channel = peer.channel;
#endif
      if (n < maxPeers) {
        memcpy(peers[n].mac, mac, 6);
        peers[n].channel = channel;
      }
      ++n;
    }
  }
  return n;
}

bool
WifiEspNowClass::hasPeer(const uint8_t mac[6]) const
{
  if (m_begin) {
    return esp_now_is_peer_exist(const_cast<uint8_t*>(mac));
  }
  return false;
}

#if defined(ESP8266)
bool
WifiEspNowClass::addPeer(const uint8_t mac[6], int channel, const uint8_t key[WIFIESPNOW_KEYLEN])
{
  if (!m_begin) return false;
  if (this->hasPeer(mac)) {
    if (esp_now_get_peer_channel(const_cast<u8*>(mac)) == channel) {
      return true;
    }
    this->removePeer(mac);
  }
  return esp_now_add_peer(const_cast<u8*>(mac), ESP_NOW_ROLE_SLAVE, static_cast<u8>(channel),
                          const_cast<u8*>(key), key == nullptr ? 0 : WIFIESPNOW_KEYLEN) == 0;
}
#elif defined(ESP32)
bool
WifiEspNowClass::addPeer(const uint8_t mac[6], int channel, const uint8_t key[WIFIESPNOW_KEYLEN], int netif)
{
  if (!m_begin) return false;
  esp_now_peer_info_t pi;
  if (esp_now_get_peer(mac, &pi) == ESP_OK) {
    if (pi.channel == static_cast<uint8_t>(channel)) {
      return true;
    }
    this->removePeer(mac);
  }
  memset(&pi, 0, sizeof(pi));
  memcpy(pi.peer_addr, mac, ESP_NOW_ETH_ALEN);
  pi.channel = static_cast<uint8_t>(channel);
  pi.ifidx = static_cast<wifi_interface_t>(netif);
  if (key != nullptr) {
    memcpy(pi.lmk, key, ESP_NOW_KEY_LEN);
    pi.encrypt = true;
  }
  return esp_now_add_peer(&pi) == ESP_OK;
}
#endif

bool
WifiEspNowClass::removePeer(const uint8_t mac[6])
{
  if (!m_begin) return false;
  return esp_now_del_peer(const_cast<uint8_t*>(mac)) == 0;
}

void
WifiEspNowClass::onReceive(RxCallback cb, void* cbarg)
{
  m_rxCb = cb;
  m_rxCbArg = cbarg;
}

bool
WifiEspNowClass::send(const uint8_t mac[6], const uint8_t* buf, size_t count)
{
  if (!m_begin) return false;
  if (count > WIFIESPNOW_MAXMSGLEN || count == 0) {
    return false;
  }
  WifiEspNow.m_txRes = WifiEspNowSendStatus::NONE;
  return esp_now_send(const_cast<uint8_t*>(mac), const_cast<uint8_t*>(buf), static_cast<int>(count)) == 0;
}

void
WifiEspNowClass::rx(const uint8_t* mac, const uint8_t* data, uint8_t len)
{
  if (WifiEspNow.m_rxCb != nullptr) {
    (*WifiEspNow.m_rxCb)(mac, data, len, WifiEspNow.m_rxCbArg);
  }
}

void
WifiEspNowClass::tx(const uint8_t* mac, uint8_t status)
{
  WifiEspNow.m_txRes = status == 0 ? WifiEspNowSendStatus::OK : WifiEspNowSendStatus::FAIL;
}
```

This is obviously a bug in the ESP32 code, but I think this is a simple work around until it gets fixed.

yoursunny, 2020-12-16:
I added an equivalent patch, but did not test on physical devices. Let me know if it breaks things. Otherwise, I'll close this issue after 7 days.

TD-er, 2020-12-16:
Thanks.
Just browsing through the changes you made and it indeed looks very similar to what I suggested (and have been running for months)
So I guess it will fix this problem.
If it doesn't I'll report again.

closed 2020-12-16 by TD-er