yoursunny/WifiEspNow #22: It does not work when esp32 connected to wifi ap

ramiws, 2023-03-18:
Sending from ESP8266 to esp32-s3 works only if the s3 is not connected to wifi. Have you tested that?

yoursunny, 2023-03-20:
I do not own any ESP32-S3 so I cannot test.

maksm, 2023-08-12:
On a related note, for me the basic example of unicast doesn't work between a mixture of ESP32 and ESP8266, while between the same type of chip the example works (ESP32 <--> ESP8266 doesn't work, while ESP32 <--> ESP32 and ESP8266 <--> ESP8266 works).

Have two boards of each type, tried every combo possible, have ran out of ideas.

yoursunny, 2024-06-15:
ESP32 Arduino is now at v3.0.1, with breaking changes.
I tested the broadcast example among three different chips.
It works between ESP8266 and ESP32-C3.

However, it doesn't work on ESP32.
It seems that WiFiScan functionality is broken for the ESP32 (xtensa), even the official WiFiScan example returns zero networks.
Since the pseudo broadcast depends on WiFiScan, it wouldn't work.